4refr0nt/ESPlorer #3: Featurerequest telnet

3s1d, 2015-01-17:
BTW: Have you ever thought about wireless development? 

LUA has telnet capabilities (https://github.com/nodemcu/nodemcu-firmware/blob/master/examples/telnet.lua). The IDE would significantly benefit from supporting this as an alternative to serial.

Think about it,
Juergen

pastukhov, 2015-02-09:
Да, да! 
Было бы круто, появится возможность освободить еще 2 пина для использования.

raz123, 2015-05-11:
:+1: go telnet go!

ghusson2, 2015-09-20:
I confirm this would be nice.
Anyway, you can achieve that with telnet.lua and on Linux with : 
su -
apt-get install sudo screen
LUSER="<mon identifiant utilisateur>"
adduser $LUSER sudo
adduser $LUSER dialout
reboot
sudo apt-get install socat
sudo socat pty,link=/dev/ttyUSB2,raw,user=root,group=dialout,mode=0660 tcp:<IP, ex : 192.168.1.110>:23

and use ESPlorer on /dev/ttyUSB2 !

Or on windows, follow : http://nutandvoltz.blogspot.fr/2015/02/esp8266-remote-lua-console.html

devyte, 2016-02-08:
Hi, I'm wondering about this as well, I don't think it should be too hard to implement, something along the lines of opening a socket instead of a serial port, and routing all the commands to that.
I currently have both a lua server (aka telnet server in the examples) and an http server running at the same time on the nodemcu firmware (see devyte/nodemcu-platform). Having ESPlorer connect to the lua server would be great.