4refr0nt/ESPlorer #12: Bundle .jar as .app for Mac

phaseOne, 2015-04-18:
It's [pretty easy](http://docs.oracle.com/javase/7/docs/technotes/guides/jweb/packagingAppsForMac.html). Plus it negates the need for a folder since everything is contained.

alexglow, 2015-06-14:
+1 to this!

I've got NodeMCU loaded on the ESP-01 but am getting an error, "cannot open init.lua", and it sounds like using ESPlorer is a good way to handle that. However, I'm on a Mac and no great shakes with Java :)

Thanks @phaseOne for the instructions – I'll try doing that for myself.

marcelstoer, 2015-12-02:
How would double-clicking a `.jar` in the Finder (or single-click in Spotlight) be different from double-clicking an OS X app bundle?

Besides, when #15 is merged this issue is resolved as well.