4refr0nt/ESPlorer #18: NodeMCU firmware autodetection

ghusson2, 2015-09-20:
Communication with MCU...
Got answer! AutoDetect firmware...
Can't autodetect firmware, because proper answer not received.

I can't understand this. With latest NodeMCU firmware (nodemcu_float_0.9.6-dev_20150704.bin), the answer to "CR/LF" is "CR/LF/> ". Which is detected by code "if (data.contains("\r\n>")) {". But I can't find where "CR/LF" is sent to the NodeMCU... Any idea to improve ?

4refr0nt, 2015-09-20:
See https://github.com/4refr0nt/ESPlorer/blob/master/ESPlorer/src/ESPlorer/ESPlorer.java#L8996

ghusson2, 2015-09-20:
Ok, I parsed your code, and I can't figure out the problem. Linux debian 8 AMD64 + NodeMCU v2 (ESP-12e) = firmware autodetection failure :)