4refr0nt/ESPlorer #30: send a ctrl-c (micropython)

ideal2545, 2016-05-18:
hi, I can't seem to figure out how to send a ctrl-c to terminate a running process, how can I do this?

4refr0nt, 2016-05-18:
For terminate sending, please press `Send To ESP` button again

ideal2545, 2016-05-18:
No luck, it doesnt terminate running proccess - is there a way to create a snippet to send "CTRL-C" ?

4refr0nt, 2016-05-18:
sorry, no

ideal2545, 2016-05-19:
Thanks for the prompt response! I'm loving this great app. Please consider adding that functionality in the future!

Thx
Jon

closed 2016-05-19 by ideal2545

fbraghiroli, 2017-08-21:
Hi,
I found a workaround to send "ctrl-c" escape sequence by saving the '\x03' char to a free 'snippet' slot.
Steps (tested on ubuntu linux)
1) open text editor (gedit)
2) press ctrl+shift+u and type 03 (this produces a single char that has 0x03 as hex value)
3) copy this char to a new snippet
4) each time you have to send ctrl-c, simply call the snippet just saved