4refr0nt/ESPlorer #32: micropython: uploading file problems/fixes.

deathmock5, 2016-06-07:
- Problem: The editor uploads items too fast, setting dummy mode to delay of 100ms solves, opening options and closeing options causes this value to reset.
- Solution: make delay of 100ms default.

- Problem: Characters in string are taken as literals.

- Solution: change pyfiller.java->escape to
`
        public String escape(String str) {
        char ch;
        StringBuilder buf = new StringBuilder(str.length() * 2);
        int intValue;
            for (int i = 0, l = str.length(); i < l; ++i) {

            ch = str.charAt(i);
            if (ch == '"') {
                //intValue = ch;
                buf.append("\\");
                //buf.append(ch);
                buf.append(ch);
            } else if (ch == '\'') {
                intValue = ch;
                buf.append("\\");
                //buf.append(intValue);
            } else {
                buf.append(ch);
            }
            else if(ch == '\\')
            {
            	buf.append("\\\\");
            }
            else
            {
            	buf.append(ch);
            }
        }
        return buf.toString();
        } // escape`

- Problem: file writing adds it all as a single line.
- Solution: Change line 32 of Pifiller to: ` sendBuf.add("f.write('" + escape(subs) + "\\n')");`

jreznik, 2016-07-16:
I can confirm all three issues.

formatc1702, 2016-11-26:
Can confirm, especially the file writing as a single line.

My quick and dirty fix is adding a \n to each line in the code editor for now...