4refr0nt/ESPlorer #37: Send button in console sends the command but the text remains in the input

mflmartin, 2016-07-29:
So it is very annoying to have to delete it manually before adding a new one. 

Se the attatched image. I sent "import os", but it is still in the input. So you can not write and send things fast, because you have to delete all the time.

Thanks. 

[image]

devsaurus, 2016-07-30:
I appreciate this "feature" for being able to repeat this command in the buffer whenever it's needed.

For copy & exec a sequence of commands just select them in the editor on the left and press <Alt>-B.

4refr0nt, 2016-08-08:
Please, try snippets - its more powerfull. Command line for fast inline coding only.