4refr0nt/ESPlorer #41: ESplorer v0.2.0-rc5 no longer autodetects firmware

gknauf, 2016-08-30:
I'm using ESP-01 modules with an [USB-CH340 adapter](http://plaintron.onlinetoolbox.de/ESP01USB-CH340_02-600x480.jpg); since v0.2.0-rc5 (binary downloaded as v0.2.0-rc6 from your site, but title bar and About dialog names it v0.2.0-rc5) the autodetection of the AT firmware of my ESP-01 modules does no longer work; when I switch back to v0.2.0-rc3 then my modules are properly detected again; modules have:
AT version:0.40.0.0(Aug  8 2015 14:45:58)
AT version:0.60.0.0(Jan 29 2016 15:10:17)

dunk8888, 2016-10-13:
doesnt detect firmware of nodemcu 0.9 eather nore does the r6 version,it comes up with flash errors Communication with MCU..Got answer! Communication with MCU established. AutoDetect firmware...  AT-based firmware detected. read: r=1 at 4070D000 ERROR in flash_read: r=1 at 4070E000 ERROR in flash_read: r=1 at 4070F000
flash_read: r=1 at 4070F001 and so on and so on

dunk8888, 2016-10-13:
its serial problem i think,i only get errors output when i change baurd rate to 9600 or i get nothing?? please help

dunk8888, 2016-10-13:
but works fine on r2 version of esplorer

oddstr13, 2017-04-28:
Does not detect Micropython either. This is true for both trunk and the rc5/rc6 binary.
[image]
The random characters you see is the ESP boot log, at a different baud rate.

closed 2017-04-29 by 4refr0nt