4refr0nt/ESPlorer #43: Sending garbage in strings to NodeMCU v1.0

Jflima92, 2016-09-22:
I really don't know what is happening, but when I send or save to nodemcu, lua code, most of the times it can't send and throw stdin errors and in the console I see that what is being sent contains garbage characters and that's why it is breaking. Any ideas on how to fix this? 

Example: Sending print("Hello World") -> print("\"#$...") and other garbage.

df6dbf, 2017-04-13:
Same behavior here with Micropython v1.8.7 (stable). Works most time with Code abt 40 Lines lenght. Longer code is corrupted. Problem occurs on several Machines with several Operating Systems. Playing with the FIFO-Buffers in the Device-Manager doesnt help.

Kind regards!

Holger

oddstr13, 2017-04-28:
This sounds like an issue with the USB-serial chip and the ESP having inaccurate baud rades.
I would suggest trying lower baud rates, as the relative error would get smaller then.

closed 2023-09-29 by jfmlima