4refr0nt/ESPlorer #48: GUI bug driving me crazy :)

jeanfbrito, 2016-12-16:
If the tooltip is summoned one time, all the GUI get weird and each button redraws the window. Take a look:
[image]
Any way to disable tooltips?

4refr0nt, 2016-12-16:
Sorry, but you can disable each tooltip only in source code

jeanfbrito, 2016-12-16:
Yes, I'm installing the NetBeans to do that. Any idea of what is causing that bug?

4refr0nt, 2016-12-16:
I can't reproduce this bug

jeanfbrito, 2016-12-16:
There is a bug with multiple monitors, I think that is it. Do you use more than a monitor?

4refr0nt, 2016-12-16:
no

jeanfbrito, 2016-12-16:
HA! Take a look!
All the tooltips are working well, and they are yellow:
[image]
But, the AT commands line tooltip is of another cooler, gray, and ONLY WHEN this tooltip is evoked, that the interface bugs.
[image]

If I remove the text from it, the bug stops. Then, there is a problem with this component.
Is there something so different in it? The "send" button, on the right, have this problem too.

jeanfbrito, 2016-12-16:
The color changes when the component is disabled. The color is not the problem. The problem is on the "command" to send area.

jeanfbrito, 2016-12-16:
Just removing the "Command to send" tooltip resolved my problem. Weird not?

4refr0nt, 2016-12-16:
Its may be UI library bug

jeanfbrito, 2016-12-16:
Not fully solved, sometimes when the tooltip of a "empty tooltip" apears with just an spance inside, bugs the interface again. This is driving me nuts... Haha

oddstr13, 2017-04-26:
Recording of the issue appearing.
[image]

Hovering the input bar breaks the UI.
Moving the ESPlorer window to the secondary monitor gives a working UI on that monitor, until the hover-text of the command input is displayed again. Moving back to main monitor does not fix the issue.

OS: Windows 10
GPU: GTX1070
Java:
```
java version "1.8.0_121"
Java(TM) SE Runtime Environment (build 1.8.0_121-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.121-b13, mixed mode)
```

The following appears on the console, not sure if related:
```
[0x7FFEC09E5110] ANOMALY: meaningless REX prefix used
[0x7FFEB6D84250] ANOMALY: meaningless REX prefix used
```