4refr0nt/ESPlorer #49: How does the autodetect work?

jhoogeboom, 2017-01-17:
I'm having a bit of trouble getting ESPlorer to autodetect my NodeMCU firmware, what is the expected handshake? It seems to be a caret distance of more than 10? The thing is, this only seems to happen when the device is restarted, not when opening the port. So when opening the port it just hangs unless resetted. Unless autodetect is disabled, then it works fine. I'm a bit new, in earlier versions upon opening the port it printed out the firmware version?

public void CheckComm() {
        if (!AutodetectFirmware.isSelected()) {
            portJustOpen = false;
            return;
        }
        portJustOpen = true;
        TerminalAdd("\r\nCommunication with MCU..");
        checker = new ActionListener() {
            public void actionPerformed(ActionEvent evt) {
                if (portJustOpen) {
                    btnSend("");  // only CR+LF
                    TerminalAdd(".");
                    if (Terminal.getCaretOffsetFromLineStart() >= 10) {
                        TerminalAdd("\r\n");
                    }
                } else {
                    try {
                        openTimeout.stop();
                    } catch (Exception e) {
                        log(e.toString());
                    }
                }
            }
        };
        openTimeout = new Timer(3000, watchDog);
        openTimeout.setRepeats(false);
        openTimeout.setInitialDelay(3000);
        openTimeout.start();
    }

4refr0nt, 2017-01-18:
```
                      if (data.contains("\r\n>>>")) {
                            TerminalAdd("\r\nMicroPython firmware detected, try get version...\r\n\r\n");
                            btnSend("import sys; print(\"MicroPython ver:\",sys.version_info)");
                            LeftTab.setSelectedIndex(0);
                            SetFirmwareType(FIRMWARE_MPYTHON);
                        } else if (data.contains("\r\n>")) {
                            TerminalAdd("\r\nNodeMCU firmware detected.\r\n");
                            btnSend("=node.heap()");
                            LeftTab.setSelectedIndex(0);
                            SetFirmwareType(FIRMWARE_NODEMCU);
                        } else if (data.contains("\r\nERR")) {
                            TerminalAdd("\r\nAT-based firmware detected.\r\n");
                            btnSend("AT+GMR");
                            LeftTab.setSelectedIndex(1);
                            RightExtraButtons.setVisible(false);
                            RightSnippetsPane.setVisible(false);
                            NodeFileManagerPane.setVisible(false);
                            FirmwareType = FIRMWARE_AT;
                        } else {
                            TerminalAdd("\r\nCan't autodetect firmware, because proper answer not received (may be unknown firmware). \r\nPlease, reset module or continue.\r\n");
                        }
```
We expect lua console `\r\n>` after firmware start.

jhoogeboom, 2017-01-19:
Thanks, that makes sense, any idea why ESPlorer is not finding the \r\n> ? screen /dev/tty.SLAB_USBtoUART 115200 yields a \r\n> just fine when presented a CR+LF

4refr0nt, 2017-01-19:
not all version NodeMCU firmware has identical start screen. Early, all version has answer `\r\n>` on `\r\n` (empty command), but now not.

closed 2017-01-19 by 4refr0nt

jhoogeboom, 2017-01-19:
This is on the most recent build of NodeMCU 1.5.4.1, both ESPlorer and screen were used with the same version. ESPlorer hangs, screen does not.

4refr0nt, 2017-01-19:
At now many version of NodeMCU with different answers. I can't test ALL versions for his answers, but if you know what you doing then you cant make new PR for properly autodetect your version of NodeMCU.

jhoogeboom, 2017-01-20:
Sorry I'm not sure where it's going wrong yet, and I understand you can't test all versions. Do you know  of a known working NodeMCU firmware version?