4refr0nt/ESPlorer #52: Seems lost characters during Save to ESP

whc2001, 2017-04-01:
> file.remove("test.lua");
> file.open("test.lua","w+");
> w = file.writeline
w([==[str = "Hello World!";]==]);
w([==[print(str);]==]);
file.close();
dofile("test.lua");
stdin:1: attempt to call field 'writelinew' (a nil value)
stack traceback:
	stdin:1: in main chunk
> 

Each time I hit "Save to ESP" the console returns something similar to this. However "Send to ESP" works. No use changing baud rate.

4refr0nt, 2017-04-01:
Try this `\[` and `\]`

whc2001, 2017-04-02:
But I didn't write square brackets in the test program. The code with square brackets above was added automatically by ESPlorer.

tobyliu415, 2017-08-11:
I found this problem my cased by settings -> send -> dumb mode -> delay = 0
try to change the delay time of dumb mode.

21iczhongli, 2018-08-20:
I solved the same problem by reseting the dump mode of ESPlorer

alex85k, 2018-12-02:
Dumb mode was also an issue in my case. It is switched on automatically for micropython but not switched back when switching to NodeMCU.