4refr0nt/ESPlorer #55: Send settings does not stick. (MicroPython)

oddstr13, 2017-04-28:
In MicroPython mode, Send settings does not stick, and gets reset to "Dumb Mode" with 0ms line delay.
This happens uppon hovering the "Select firmware -> MicroPython" checkbox/text, and after saving a file.

If I change the setting, and make sure not to hover "MicroPython" when changing to the scripts tab, I am able to upload a file, before the settings are reset and I have to change them again.

Could be related to #41, and magic happening there.

[image]

You can also clearly see #48 happening here, when hovering the Cancel button.

mickley, 2017-12-21:
@oddstr13: I was able to make a temporary fix by adding LineDelay.setValue(100); to pySaveFileESP()

This sets the line delay in dumb mode to > 100ms every time you try to send a file, slow enough to work.

Josverl, 2018-01-03:
I have been working on improving support for uPython. This has not yet been merged into this repo , but you can try the fork located at : https://github.com/Josverl/ESPlorer/tree/MT-Build to see if that helps resolve your issues. In that version I have re-written the file upload handling as well as added other uPython functionality (as I had similar issues ). I now have a stable upload on 115200 with no additional need for delays or timings.