4refr0nt/ESPlorer #63: Crash when press Open for serial connection

netstrike, 2017-10-13:
Today... when i have installed the JDK 9.0 ESplorer crash everytime ....

[hs_err_pid5988.log](https://github.com/4refr0nt/ESPlorer/files/1382228/hs_err_pid5988.log)

4refr0nt, 2017-10-13:
Its problem jSSC-2.8_x86_64.dll jSCC library

d-rez, 2017-10-23:
Can confirm. Would you be able to fix this or do we have to roll back to JDK8?

RouterBox, 2017-10-28:
Confirming Esplorer is broken on windows 10 with jdk 9.

4refr0nt, 2017-10-28:
I can't fix jSCC library, try roll back to JDK8

aliezzzz, 2017-11-08:
I did roll back to JDK8u151,but it still crashed,when I press Open for serial connection

Tanonic, 2017-12-22:
Having the same problem on win 10 with jdk 9

RouterBox, 2017-12-25:
Roll back to JDK 8.  Fixed it for me.

d-rez, 2018-06-23:
@4refr0nt, just a heads-up that the same behaviour is exhibited by Java 10. Since it's been broken for 2 Java versions would you be able to work around this somehow?

fribeiro1, 2018-12-21:
Same for Java 11.

d-rez, 2018-12-22:
I guess the dev doesn't care anymore, but you can install jdk8 alongside newer ones and then just modify the path to Java executable in the .bat file to point towards jdk8 directly

davidhbrown, 2019-04-08:
In case it helps anyone, here's a modified batch file that targets the current-as-of-today JRE 8. The included batch file wasn't working for me anyway until I included the "%~dp0\" to give Java the full path to the jar.

    "C:\Program Files\Java\jre1.8.0_201\bin\java.exe" -jar "%~dp0\ESPlorer.jar"

kabilramar, 2019-04-23:
Mentioned the above absolute path but it didn't works what is the reason

davidhbrown, 2019-04-23:
@kabilramar, Well, that was a couple weeks ago... JRE 8 is now up to version 211 and/or 212, so your best bet would be to poke around in your Program Files\Java folder and see which version you do have.

kabilramar, 2019-04-24:
Got it when I down graded to jdk 1.8

kabilramar, 2019-04-24:
Thank you and what are the firmwares to be downloaded for flashing in nodemcu

magixyu, 2019-10-29:
I replaced the jssc with the one from https://github.com/java-native/jssc/releases, then ESPlorer works with JDK 12/13.

jhfoo, 2020-06-04:
There must be a long-term alternative to ESPlorer. This app is outdated and it's not reasonable to continue using above workarounds.

RouterBox, 2020-06-04:
@Foo JH

Check out:

http://chilipeppr.com/

jhfoo, 2020-06-05:
Thanks @RouterBox. The folks who wrote chilipeppr are very smart. Love the automation built in, GitHub/ Cloud9 integration, and customisable widgets!

Qiguang, 2020-07-16:
I just used java 8, but the problem is still there.
I downloaded and installed jdk-8u261-windows-x64.exe on my win10 laptop.
And edited JAVA_HOME to C:\Program Files\Java\jdk1.8.0_261
But when I click the "OPEN" button on esplorer, It crashed
[hs_err_pid13752.log](https://github.com/4refr0nt/ESPlorer/files/4932371/hs_err_pid13752.log)

rwlakin, 2020-11-19:
I downloaded and installed jdk-8u271-windows-x64.exe on my win10 laptop.
C:\Program Files\Java\jdk1.8.0_271
ESPlorer crashed when I click the "OPEN" button

rwlakin, 2020-11-19:
davidhbrown
I like your idea using batch file but I need help using it.
"C:\Program Files\Java\jre1.8.0_201\bin\java.exe" -jar "%~dp0\ESPlorer.jar"

olebed, 2020-11-19:
Hi David!

Doesn't clicking on ESPlorer.bat work?
It might not work if you do not have "java.exe" in your PATH. If you do not
(simply type "java.exe" in cmd window to check) and do not want to modify
your PATH you can
replace java.exe with "C:\Program Files\Java\jre1.8.0_201\bin\java.exe" in
the ESPlorer.bat.
The part *-jar "%~dp0\ESPlorer.jar"* looks weird from the syntax point of
view.

Assuming that *jar* file is located in the same folder as *bat* you need
something like this:

"C:\Program Files\Java\jre1.8.0_201\bin\java.exe" -jar ESPlorer.jar

in the ESPlorer.bat

Regards,
Oleksandr

olebed, 2020-11-19:
> Still working on my ESPlorer, modified the .bat file, still crashes
> Here's what's in my ESPlorer.bat file
> "C:\Program Files\Java\jre1.8.0_271\bin\java.exe" -jar ESPlorer.jar

Hi, there is not enough information to help you.
Please send a stacktrace (error messages).
If you do not see them then open windows terminal by running "cmd",
change directory via "cd" to the one where ESPlorer.jar is located and run

*"C:\Program Files\Java\jre1.8.0_271\bin\java.exe" -jar ESPlorer.jar*

You should see error messages in the window after the program crashes.
Please post them here.
--
Oleksandr

rwlakin, 2020-11-20:
hi olebed
I have running esplorer.bat, original is   java -jar "ESPlorer.jar" 
When I select "open" both ESPlorer and cmd crash
I tried *"C:\Program Files\Java\jre1.8.0_271\bin\java.exe" -jar ESPlorer.jar* this in cmd prompt; just gives this:   "The filename, directory name, or volume label syntax is incorrect."  Probably because I don't know what I'm doing.  I need more help with this.

abhishekkarigar, 2020-11-24:
Hi Folks,
faced the same issue, instead of changing my java version , i downloaded the source code ,
build the maven project , then use the jar from target folder.  

**https://github.com/4refr0nt/ESPlorer**  
  
> java -jar target/Esplorer.jar  

voila .... :)

olebed, 2020-11-24:
Hi abhishekkarigar,
thanks a lot for sharing this!

A few questions to you:
1. What OS are you using?
2. What is your java version? Run "java -version" in the terminal
3. Could you check if the pre-build application works for you too? Download
the one from "releases" on the github page.

Thanks!

rwlakin, 2020-11-25:
Thanks abhishekkarigar
I downloaded the release version Esplorer 0.2.0 @ github and it works with no crashes.  In the process of using it.
Roger

abhishekkarigar, 2020-11-26:
@olebed 
1. Windows 10
2. java version "1.8.0_271"
3. yes pre-build application works now too :)

olebed, 2020-11-26:
Nice to know abhishekkarigar!
Thanks! :-)

tgrotowski, 2021-03-15:
Guys, There is some solution for this crash? 

My system win 10 x64. I tested jre1.8.0_281. I got crash when click "open". App dowloaded from here: https://esp8266.ru/esplorer/#download (zip version)

Or maybe guys do you know some alternative software?

abhishekkarigar, 2021-03-15:
@tgrotowski 
https://github.com/4refr0nt/ESPlorer/releases/tag/v0.2.0 use this,

tgrotowski, 2021-03-15:
@abhishekkarigar  it works! Thank you so much. Have a nice day! ;]

gicheve, 2022-11-20:
Thanks, it works but vertical scroll cannot be moved down and files on right are not visible if they are more than 7-8...

vecesneni, 2023-09-22:
For me this v0.2.0 version still crashes on opening the port due to the outdated jssc library. The Azul Java support updated for me the jssc in the jar file to the latest, jssc-2.9.5, I have attached. Please use this, if you have problems.
[ESPlorer.zip](https://github.com/4refr0nt/ESPlorer/files/12702172/ESPlorer.zip)