4refr0nt/ESPlorer #67: Save to ESP doesn't work for Micropython

alexzaitsev, 2017-12-27:
I've written a file using 'Save to ESP' button.
[image]
It was successful. 
Then I checked the file using `picocom`:
[image]
And seems that it's empty.

Btw, several times I couldn't even write a file using 'Save to ESP' - the last line was cut.

alexzaitsev, 2017-12-27:
**UPD**
After several tries seems that it wrote file but still it was done not properly:
[image]

alexzaitsev, 2017-12-27:
That's because you use `f.write(...)` and don't add newline symbol.

Josverl, 2018-01-03:
I have been working on improving support for uPython. This has not yet been merged into this repo , but you can try the fork located at : https://github.com/Josverl/ESPlorer/tree/MT-Build to see if that helps resolve your issues. In that version I have re-written the file upload handling as well as added other uPython functionality.

kurtsvl, 2018-02-11:
Josverl i use your version of Esplorer for ESP32 programming and it works great unlike the original, the binary mode of loading is great, thank you. P/S I hope the author uses your changes in The original version ESplorer.

liudr, 2018-03-26:
I got the same issue using MicroPython. I think the IDE strips line ends and never puts them back when writing to the dev board. This should be an easy fix (?) by the author.