4refr0nt/ESPlorer #68: Edit and download lua file from esp8266 does not work with API 2.1

jwmeyer, 2018-02-02:
When using the right click menu on the lua file in the esp8266 the system hangs and does not create a new  script or a save file window.
When connecting an esp8266 with API 2.0 or lower this functionality works correctly.

StKob, 2018-02-05:
I will look into this. Could you tell me exactly which firmware you have on esp?

jwmeyer, 2018-02-24:
**I guess I was mistaken. On a wemos D1 mini with this firmware:**
NodeMCU custom build by frightanic.com
	branch: master
	commit: 443e8219527f5b2190324a969a4586f9d3d731bf
	SSL: false
	modules: bit,enduser_setup,file,gpio,i2c,mqtt,net,node,rotary,tmr,u8g,uart,wifi
 build 	built on: 2017-12-10 14:20
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
**It works correctly but with another board connected with a sparkfun FTDI the edit command hangs until timeout.**
NodeMCU custom build by frightanic.com
	branch: master
	commit: 5073c199c01d4d7bbbcd0ae1f761ecc4687f7217
	SSL: false
	modules: file,gpio,mqtt,net,node,pwm,tmr,uart,wifi
 build 	built on: 2018-02-01 21:17
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

jwmeyer, 2018-03-08:
The D1 mini no longer works. Not sure what changed.