4refr0nt/ESPlorer #70: Themes/Skins

otalado, 2018-03-04:
I have great request to make:
because of my **visual impairment** (not so uncommon at my age) it is very hard to me to use ESPlorer with available skins because they are all **dark themes** (light text on dark/black background). Also current GUI colours (black on grey) are very hard (almost harder)on my eyes. Because of that I'd really to request that you provide at list one **light coloured skin** (i.e. dark text on white background). I'd request **white background** for the GUI as well if that is not overly difficult.
If that means too much effort for you - is it possible that I do it myself somehow?? Of course I'd need some instructions how to go about it.

LuisPeregrina, 2020-02-01:
@otalado do you use the High Contrast setting in your OS?

otalado, 2020-02-02:
No, I don't use HighContrast. Should I ?