4refr0nt/ESPlorer #74: Nothing happens on "Open" connection

kalanaw, 2018-08-18:
**Version:** ESPlorer v.0.2.0-rc5
**OS:** Mac OS 10.13.6
**Board:** ESP8266

I've flashed with default firmware via nodemcu-build.com. I'm able to Connect to board through CoolTerm and also use "NodeMCU-Tool".

However when I click "Open" on ESplorer it says "Communication with MCU" and nothing happens.
Any guesses what could go wrong?

[image]

anudeepsharma254, 2021-02-06:
I'm also facing same problem with ESplorer I don't know whether is due to ESP32 or due to Esplorer because When I tried with other ESP32 also not resolved  please suggest me way to get out of this error
[image]

hinsech475, 2021-02-06:
Hi Anudeep.

i will be honest with you, i gave up on using ESPlorer because I never had any assistance to resolve my problem.

Nonetheless, as I said in my comment of August 18th 2018, " ESPlorer would be an ideal solution if it worked as expected."

My interest was more in using Forth with the ESP8266 (Punnyforth) and I was hoping to get ESPlorer working with Lua first and adapt it tu use Punnyforth. I never did it.

I would still like to use ESPlorer with Forth but I found a usable alternative with XShell for simple Forth fun programming.

Regarding the garbled screen problem with ESPlorer, I have the impression that it starts operation with a baud rate of 115200 but switches to a different baud rate after startup.We probably just need to find the proper baud rate to use after startup.

If you find the proper baud rate, let me know.

We need luck and patience with freeware on the Internet...

Good luck... and lots of patience.

Christian Hinse

[Screenshot (48)]<https://user-images.githubusercontent.com/66728239/107116071-fd5c4d80-6896-11eb-8065-df479fa22b1a.png>