4refr0nt/ESPlorer #85: Using the jdk 11 version, after opening the com port in the software, the application flashback

hyzgit, 2020-04-23:
Using the jdk 11 version, after opening the com port in the software, the application flashback, using the jdk1.8 version is normal, but version 1.8 can not fit the high-dpi, button, the text will be very small

Qiguang, 2020-07-16:
I used 1.8 version, but still flashback when I try to open com port by clicking the "OPEN" button.
the err_log say:
_#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00000000001fb5db, pid=9716, tid=0x0000000000000698
#
# JRE version: Java(TM) SE Runtime Environment (8.0_261-b12) (build 1.8.0_261-b12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.261-b12 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug._

anshulkatta, 2020-07-25:
me too getting the same error

BoresXP, 2020-08-09:
As a workaround you can force ESPlorer to run on 32bit VM:
1. Install latest 32 JRE.
2. Open cmd and check which version you have in path: `java -version`. If you had 64 bit JRE installed before most likely you will get 64bit version by default: 
```
C:\NodeMCU\ESPlorer>java -version
java version "1.8.0_261"
Java(TM) SE Runtime Environment (build 1.8.0_261-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.261-b12, mixed mode)
```
3. In that case you have to specify full path to 32bit java in ESPlorer.bat:
```"C:\Program Files (x86)\Java\jre1.8.0_261\bin\java.exe" -jar "ESPlorer.jar" ```
4. Done. Just run ESPlorer.bat as before. Port will open without exception.